OSUPCVLab/mobile-ar-sensor-logger #1: Testing issues

Nronaldo, 2020-06-23:
Dr. Huai
Thanks for your excellent job.
I collected images and imu measurement with your logger on my Xiaomi Mi 8. I found I cant lock the AF by tapping the screen. I also solved the measurement with Vins-mono. It seems that we should be very careful about the imu parameters to provide positioning  results with reliable scale.

JzHuai0108, 2020-06-23:
Huai Yu also reported a similar issue about a Xiaomi phone of CC9 model with MIUI 11.0.3 OS (based on Android 10), i.e., the focal length was not fixed after tapping the screen.
Unfortunately a Xiaomi phone is not accessible to me yet, and I don't think there is an corresponding Android emulator. So I have to postpone solving this issue. 
That said, by browsing the movie_metadata.csv recorded by your Mi 8 phone, I don't see much change in the focal length even when the AF is not locked, e.g., 958.0559 at the end vs. 957.1534 at the start.

I agree that in running VIO algorithms, the IMU noise parameters have to be set with care. Often times, a manual tuning is needed. From my personal experience, I often start tuning a VIO method with below IMU noise parameters for visual inertial data collected by smartphones,
```
    sigma_g_c: 24.0e-4 # gyro noise density [rad/s/sqrt(Hz)]
    sigma_a_c: 16.0e-3 # accelerometer noise density [m/s^2/sqrt(Hz)]
    sigma_bg: 0.03 # gyro bias prior [rad/s]
    sigma_ba: 0.1 # accelerometer bias prior [m/s^2]
    sigma_gw_c: 2.0e-5 # gyro drift noise density [rad/s^s/sqrt(Hz)]
    sigma_aw_c: 5.5e-5 # accelerometer drift noise density [m/s^2/sqrt(Hz)]
```
For VINS-Mono, the above values are translated to 
```
acc_n: 1.6e-2 * sqrt(50) 
gyr_n: 2.4e-3 * sqrt(50) 
acc_w: 5.5e-5 * sqrt(50)
gyr_w: 2.0e-5 * sqrt(50)
```
where 50 Hz is the IMU sampling rate.

zzbwhu, 2020-09-26:
你好
很优秀的工作，收获很大
我遇到了一个比较小的问题
从github下载之后打开工程
Sync 之后出现了一个  path may not be null or empty string. path='null'
                                    Open File
的错误  
点击  Open File 之后  跳转到了  这里   storeFile file(properties.getProperty('storeFile'))
不知道这是什么原因呢  应该怎么解决呢

JzHuai0108, 2020-09-26:
The reason is that you do not have the keystore and key which are required for building the Android project in release mode.
Two solutions:
*The simple one* 

Delete the release section from [app/build.gradle](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/build.gradle#L12-L19).

*The hard one*

1. [generate an upload key and keystore](https://developer.android.com/studio/publish/app-signing#generate-key) and then 
2. put details of the key and keystore in android-project/local.properties and it will look like
```
ndk.dir=****
sdk.dir=*****
keyAlias=****
keyPassword=****
storeFile=****
storePassword=****
```
3. Now the project should be able to be built in either debug or release mode.

zzbwhu, 2020-09-26:
> The reason is that you do not have the keystore and key which are required for building the Android project in release mode.
> Two solutions:
> _The simple one_
> 
> Delete the release section from [app/build.gradle](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/build.gradle#L12-L19).
> 
> _The hard one_
> 
> 1. [generate an upload key and keystore](https://developer.android.com/studio/publish/app-signing#generate-key) and then
> 2. put details of the key and keystore in android-project/local.properties and it will look like
> 
> ```
> ndk.dir=****
> sdk.dir=*****
> keyAlias=****
> keyPassword=****
> storeFile=****
> storePassword=****
> ```
> 
> 1. Now the project should be able to be built in either debug or release mode.

because the hard one is very complex , so i adopt the simple one, and it works well 
thank you

closed 2020-09-27 by JzHuai0108