OSUPCVLab/mobile-ar-sensor-logger #3: Vertical and horizontal placement

Bing1002, 2020-10-20:
Dear Dr. Huai, 

I have a question about how we should hold smartphone, vertically or horizontally? When I vertically hold smartphone, the trajectory result seems ok, but when horizontally hold, the results is not correct. Should we change any configuration in config file or this does not affect performance? Thank you.

JzHuai0108, 2020-10-20:
Please help me understand the issue fully.
1. Can you explain to me what phone model, what phone OS are you testing with?
2. Also, try to rotate the phone from a vertical orientation to a horizontal one, in the meantime,
do you see the app disappears and appears again, i.e., the app restarts?

Bing1002, 2020-10-20:
Hi Jianzhu,

Thanks for your reply. I tested it on both iPhone se and iPod touch and they’re good for holding device vertically. I mean will it be anything different between vertically holding and horizontally holding. If this does not affect data collection and process, I will collect more data horizontally and see if they are good to process. Thank you.

JzHuai0108, 2020-11-03:
I have run across the app recreation problem only on Android phones, and have fixed it by locking the display orientation.
Since you are working with an iOS device, this should not be the cause of your problem.
After all, the major cause probably lies in the program you used to process the data.
Let me know if you have fixed your problem.

closed 2021-05-05 by JzHuai0108