OSUPCVLab/mobile-ar-sensor-logger #4: User Defined IMU Frequency

gokberkgul, 2020-10-29:
In the android version of the code, under IMUManager class, I saw that you are using
`private final int mSensorRate = SensorManager.SENSOR_DELAY_GAME;`
as the default delay for the IMU data acquisition which would be around 50 Hz. It would be really good to let the end-user define the frequency of the IMU data. Sensor rate can be defined as an integer in microseconds, which I did and build the app again. I have no Android development experience, otherwise, I would have added it myself. Thank you for this amazing app.

JzHuai0108, 2020-10-30:
Absolutely. The user should be able to specify the sampling rate of IMU.
I will add an android interface for adjusting parameters of recording as soon as I can.

JzHuai0108, 2020-11-20:
The feature is done, but I am retouching the UI and running further tests to improve its mean time between failures.

JzHuai0108, 2020-12-27:
Finally life gives me some breathing time, and I test the app on a couple of android phones. It works without any problem.
Please check out the [latest android release](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/releases).
Your comments are welcome.

closed 2021-05-05 by JzHuai0108