OSUPCVLab/mobile-ar-sensor-logger #6: Pixel alignment issues

anastasiia-kornilova, 2020-11-08:
Dear authors, thank you for such a useful app! That is what every sensors lab dreams about.

Investigating your code and paper, I don't understand the thought about the disadvantages of using ImageReader and MediaCodec together; using GL and MediaMuxer seems to be overwhelmed because it generates additional load on the processor in comparison to optimized MediaRecorder. In the paper, there is a statement, "To efficiently display the camera frames on the screen, the GLSurfaceView is used for rendering. Additionally, it shares the rendered frame with a MediaCodec encoder, thus avoiding the pixel alignment issues which arise in using ImageReader and MediaCodec together." Could you give more explanations on what is "pixel alignment issues"?

JzHuai0108, 2020-11-10:
We have considered several approaches to efficiently display and record images. The below are my notes in developing the program. It deserves space in the submitted paper.

Methods to save camera frames in Android
1. OpenCV4Android: hard to get image timestamps
2. Camera API: Camera PreviewCallback, onPreviewFrame, are deprecated.

For camera2 API, 

|----------------------------|-----------------------------------|----------------------------------|-------------------------------------------|
| camera 2 API | Frame camera characteristics and timestamps |    Efficiently record frames |  Coding complexity  |
| 3. MediaRecorder   |                  No                   |   Yes                               |  Low        |
| 4. ImageReader + saving images |  Yes             | Not efficient, slow I/O    |   Low              |
| 5. ImageReader + MediaCodec + MediaMuxer |   Yes |  Efficient  | High because it is error prone to fiddle with chroma planes, for instance, COLOR_FormatYUV420Flexible may be used by MediaCodec while YUV_420_888 is used by ImageReader. see the [issue1](https://stackoverflow.com/questions/18795353/using-mediacodec-to-save-series-of-images-as-video) and [issue2](https://stackoverflow.com/questions/13703596/mediacodec-and-camera-colorspaces-dont-match). |
| 6. GLSurfaceView + MediaCodec + MediaMuxer | Yes  | Yes  | Medium |

I hope you can see why we choose approach 6 in implementing the program and what we mean by the pixel alignment issue in using ImageReader and MediaCodec together.

anastasiia-kornilova, 2020-11-10:
@JzHuai0108 , yes, now your solution is absolutely clear. Thank you for such a detailed explanation!

closed 2020-12-27 by JzHuai0108