OSUPCVLab/mobile-ar-sensor-logger #7: IMU calib and output resolution

GabHoo, 2020-11-24:
Hi there,

I recently downloaded the app for collecting and creating some datasets and it works great but, since I am very new in the field there are still a pair of issues I need to understand and solve.

The software that I intend to elaborate my sequences with is ORBSLAM-3. In order for it to run some visual-inertial datas, a setting file is required where obviously all the parameters regarding IMU and camera are stored.
I managed to find out intrinsic parameters of my android smartphone  camera and they work fine enough on visual only dataset elaboration but when it comes to visual-inertial it also requires some kind of IMU calibration parameters (more specifically a 4x4  transformation from camera to body-frame (imu) matrix and Gyroscope and Accellerometer noise and walk. [e.g. https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/TUM_512.yaml ]
I was wondering if there was a way to find those parameters from Android the system its self (I guess that It would be the best)
or in some other way.

Moreover I wanted to ask if there was a way to change the resolution of the output video, and if not how hard would it be to implement. That would be awesome!

Thank you very much for you work!

JzHuai0108, 2020-11-24:
(1) For IMU noise parameters, I often find good results with the parameters at #1.
To my knowledge, the Android API does not contain these noise parameters, especially considering that a set of noise parameters may be suitable for a VINS method, but unacceptable for another VINS method.
That is, these VINS methods may need different noise parameters for the same IMU.
(2) For the camera IMU extrinsic parameters, the IMU and camera on literally all smartphones (Android or iOS) 
have the same nominal orientation, as shown in [the drawing](https://github.com/AaltoVision/ADVIO/issues/17).
Because the IMU and camera are not too far, within 10 cm, their relative translation can be safely ignored for practical VINSs.
Thus, their relative pose, T_imu_cam = 
[0, -1, 0, 0;
-1, 0, 0, 0;
0, 0, -1, 0;
0, 0, 0, 1]
Note point of homogeneous coordinates in imu frame = T_imu_cam * point of homogeneous coordinates in camera frame.
(3) A bit coding is needed to change the frame resolution within the app. Otherwise, many scripts can be used to downsample the resulting frames, e.g., [by tuning max_video_frame_height in this python script](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py).

closed 2020-12-27 by JzHuai0108

qamaruislam, 2021-05-09:
Dear @GabHoo @JzHuai0108 

Thank you so much for your illustration. But I am new in this area just @GabHoo like you. I am really grateful if you Kindly suggest me How I run my own dataset build by Mars Logger on android smartphone on ORB SLAM3 Because I'm little bit confused here the method is elaborate here is for ROS based implementation. I am thinking if it is possible for Non ROS based version also to running our own sequences on ORB SLAM3.

Your suggestion and recommendation are highly appreciated.

Thank you.

GabHoo, 2021-05-09:
So I did this few months ago some I might be missing something but in order for OS3 to run you need intrinsic of the camera (you can calibrate it with OpenCV), then you need IMU noises which I found online and also this Matrix which represents the rototraslation between IMU and cameras. This Matrix is a little tricky to calculate but you can perform a Calibration with _Kalibr_ that will output exactly this 4x4. ofc you also need to get the frames from Mars logger video

GabHoo, 2021-05-10:
Even thought this is not exactly the best place to do it I will try to provide some more infos but as I said, I don't remember all of it.
To calibrate the camera I used this script, provided inside OpenCV and explained here: https://docs.opencv.org/master/d4/d94/tutorial_camera_calibration.html
You just need to input various pictures of the checkboard (from different point of view) and the software will give the intrinsics bacl.
Regarding your IMU-camera calibration I suggest you to follow the video tutorial https://github.com/ethz-asl/kalibr.

qamaruislam, 2021-05-16:
Dear @GabHoo 

Thank you again for your explorative explanation. However, It is there any way to run data with mars logger without ROS version of ORB-SLAM3? It seems here only explain to run the sequence with bag file for ROS version of ORB SLAM3. How to modify the dataset file from Mars Logger to run sequences in TUM-VI or EuRoC Dataset Format?

Any suggestions and recommendation are highly appreciated.

Thank you.

Kind Regards

JzHuai0108, 2021-05-18:
I have partly answered the question in issue #16. I think it may be better to discuss in that independent issue.

GabHoo, 2021-05-19:
If you have some programming experience you can just write a quick script to get all the frames from the video (a sample of this can be found in my profile), the timestamp file is either provided by marslogger or can be achieved with such script as well. python-OpenCV is the best choice imo.
Just make sure that the timestamp file is structred exactly as the one in the dataset you want to emulate (TUM-VI o KITTI or EuRoC)