OSUPCVLab/mobile-ar-sensor-logger #8: Questions for fx and fy in movie_metadata.csv

yuweijheng, 2020-12-08:
Hello Dr Huai:

Thanks for the wonderful work! I am able to properly run the example Android datasets, regardless of the VINS algorithms using my own implementation or other open-sourced ones. However, things are not going well with the datasets collected using my Google Pixel 3a. I'd be grateful if you could kindly give me some suggestions.
-  I am aware that there are some discussions about the parameter setting in [#1](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/issues/1). However, I am unable to access that page (always the Unicorn page). Not sure if I am the only one. 
- Could we simply use the values of fx and fy provided in movie_metadata.csv file? The values of fx and fy in the example config_yaml file are approximate the values in the movie_metadata.csv file multiplied by 0.5 (due to the resolution is downsampled by 2?). For pixel3a, the values of fx and fy provided by movie_metadata.csv file are approximately 3200. By contrast, the calibrated values using Kalibr are approximately 500. For cx and cy, the calibrated values are close to half of the image size (320 and 180).
- Other than that, everything seems working normally as shown in Wiki. The sampling rates for camera and IMU are 30 Hz and 100 Hz, respectively. The timestamps look fine. The visual and inertial measurements are synchronized to the same clock. The converted bag file looks good as well. For the test scene, we walk a straight line along a corridor. Any suggestion?
Thanks.

JzHuai0108, 2020-12-09:
You are not alone in seeing unicorns as I also run into them many times.
I updated [the wiki](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki) so that you may better understand how to set the camera parameters.

yuweijheng, 2020-12-09:
Thanks for the updated wiki. I'm wondering if the values provided in movie_metadata.csv corresponds to the RAW_SENSOR size? I've successfully run some datasets using Pixel 3a and Pixel 4, but the values for fx and fy are not downscaled by 2. Take the datasets collected by Pixel 3a as an example, the values of fx and fy provided by movie_metadata.csv file are approximately 3200, while the values of fx and fy used to run the datasets are approximately 600.

JzHuai0108, 2020-12-10:
So you mean that the nominal value for fx 3200 / 2 does not work well with a VINS method.
From your case, it seems that the movie_metadata values may not be a good reference.

Here is how the focal length in units of pixels in movie_metadata is computed.
If the API supports, then it is read from the API. Otherwise, it is computed by a custom method. 
The method is implemented at [here](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95).

The comment for the function has a mistake.
It should be 
```
1/focal_length = 1/focus_distance + 1/i
focal_length is the intrinsic property of a lens, e.g., 3.5 mm.
focus_distance is the distance between lens optical center and the focusing scene, e.g., 1 m.
i is the distance between the lens and the imaging sensor. It is often very close to the physical focal length.
```

yuweijheng, 2020-12-10:
- Are you suggesting that we are able to infer the focal length from the values provided in movie_metadata.csv? 
- I am wondering if f /2 works well for your datasets collected using Honor V10 and Samsung S9+? It doesn't work well on Google Pixel phones (3a and 4). Maybe I should get a new phone...

JzHuai0108, 2020-12-10:
1. Yes. The focal length in pixels from movie_metadata.csv is usually good.
2. Yes. focal length in pixels / 2 works for our Honor V10, Samsung S9+, iPhone 6S, etc. It is strange that it does not work well with Google Pixel 3a and 4 because I expect them to support the Android API and provide accurate focal length in pixels.

yuweijheng, 2020-12-15:
I've tried some additional smartphones. It seems like the intrinsic parameters provided by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` and [here](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95) are quite different. For the smartphones I tested, it seems that the focal length in the config file should apply the values computed by [here](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95) to run the collected dataset properly. Can you kindly tell me if the focal lengths of Honor V10 and Samsung S9+ are computed by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` or [this method](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95)? Thanks.

JzHuai0108, 2020-12-16:
[The method](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95) 
first try to find the intrinsic parameters read from the field
`CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` of `mCameraCharacteristics`;
and if this is not available, it will compute these parameters based on a empirical model.

As such, if `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` provides the intrinsic parameters,
the method should give the same intrinsic parameters.

The parameters I used for Honor V10 and Samsung S9+ are read from metadata.csv except that they are downscaled by 2 because of image downsampling in the bagcreator.
For instance, movie_metadata.csv of a Honor V10 session records a focal length 979.845, I used 489.9 for the focal length in OKVIS and VINS Mono.

yuweijheng, 2020-12-16:
Thanks for the illustration. What I did in the last post was compare the focal length reported by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` and [this method](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95). It showed that the focal length reported calculated by these two methods are quite different (around 3,000 and 1,000, respectively). Just wondering if the focal lengths calculated using these two methods are similar or not on Honor V10 and Samsung S9+.

JzHuai0108, 2020-12-16:
Your observation is strange to me because I expect  the focal length reported by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` and the method to be *the same* as the method subsumes the result from `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` .
But there is a chance my implementation is buggy. I will test the app with Honor V10 and S9+ and come back to you soon.

JzHuai0108, 2020-12-27:
I 

> I've tried some additional smartphones. It seems like the intrinsic parameters provided by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` and [here](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95) are quite different. For the smartphones I tested, it seems that the focal length in the config file should apply the values computed by [here](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95) to run the collected dataset properly. Can you kindly tell me if the focal lengths of Honor V10 and Samsung S9+ in movie_metadata.csv are computed by `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` or [this method](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L64-L95)? Thanks.

Finally I got time to test Honor V10 and Samsung S9+.
Your observation is correct. The values in `CameraCharacteristics.LENS_INTRINSIC_CALIBRATION` are very different from those computed by the [snippet](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L70-L95).
For V10, 
```
2020-12-27 11:20:18.868 15667-17501/edu.osu.pcv.marslogger D/FocalLengthHelper: getFocalLengthPixel computed focal length set as (1442.060181, 1442.060181)
2020-12-27 11:20:40.350 15667-15667/edu.osu.pcv.marslogger D/FocalLengthHelper: LENS_INTRINSIC_CALIBRATION fx 0.000000 fy 0.000000 cx 0.000000 cy 0.000000 s 0.000000
```
For S9+,
```
2020-12-27 11:26:53.062 16616-16616/edu.osu.pcv.marslogger D/FocalLengthHelper: LENS_INTRINSIC_CALIBRATION unavailable
2020-12-27 11:26:53.971 16616-16899/edu.osu.pcv.marslogger D/FocalLengthHelper: getFocalLengthPixel computed focal length set as (1018.868042, 1018.868042)
```
In summary, I suspect that many android devices do not support LENS_INTRINSIC_CALIBRATION, and even it is available, the values may be unsuitable for visual odometry as you have observed.
When LENS_INTRINSIC_CALIBRATION is unavailable, the focal length in pixels in movie_metadata.csv are computed by the [snippet](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/java/edu/osu/pcv/marslogger/FocalLengthHelper.java#L70-L95).

closed 2021-05-05 by JzHuai0108