OSUPCVLab/mobile-ar-sensor-logger #9: v2.0 identifies as v1.0

IzzySoft, 2020-12-27:
Looks like you forgot to increase [`versionCode` and `versionName` in `build.gradle`](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/2.0/android-mars-logger/app/build.gradle#L26) – today's release v2.0 still identifies as v1.0 with `versionCode` 1, so it won't appear as update to Android. Could you please fix that and replace the APK at releases? Thanks!

JzHuai0108, 2020-12-28:
Thank you for the good catch. I will update it soon.

JzHuai0108, 2020-12-28:
I have updated the version, fixed a compatibility issue on Android 6.0, and updated [v2.0-android release](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/releases/tag/v2.0-android).

IzzySoft, 2020-12-28:
Thanks! My updater confirmed `versionCode: 2` and `versionName: 2.0` – so I'm closing this issue as "solved". Update will [show up here](https://apt.izzysoft.de/fdroid/index/apk/edu.osu.pcv.marslogger) with the next sync in about 4h, as usual. Thanks for fixing the issue so fast!

closed 2020-12-28 by IzzySoft