OSUPCVLab/mobile-ar-sensor-logger #10: Could you please provide the config_yaml file of the Example data?

nvgtriho, 2021-01-02:
Hello Dr Huai:
Thanks for your work. I find that the yaml file is not in the [Example data](https://drive.google.com/open?id=1AeAd4J9yW8lvAaeSxZAECQEeNQlLzoxx), so I refer to the wiki and write a yaml file. However, it doesn't work well with a VINS-fusion method as expected. The moving range of the camera is not large, but the calculated trajectory drifts far away. Could you please provide the config_yaml file of the Example data? It will be a good reference when I try out my own data. The following is part of my yaml file:

- model_type: PINHOLE
camera_name: camera
image_width: 640
image_height: 360
distortion_parameters:
   k1: 0
   k2: 0
   p1: 0
   p2: 0
projection_parameters:
   fx: 490.2406
   fy: 490.2406
   cx: 320
   cy: 180

- #imu parameters       The more accurate parameters you provide, the better performance
acc_n: 16.0e-2          # accelerometer measurement noise standard deviation. #0.2
gyr_n: 24.0e-3         # gyroscope measurement noise standard deviation.     #0.05
acc_w: 5.5e-4       # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-4       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.81007     # gravity magnitude

- body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [0, -1, 0, 0,
          -1, 0, 0, 0,
           0, 0, -1, 0,
           0, 0, 0, 1]

Thanks.

JzHuai0108, 2021-01-05:
Here is the [link](https://drive.google.com/drive/folders/1OrrODEqNn_0_sJwqaNt-faYWdzJLOhv4?usp=sharing) to the scripts and configuration files for the example data.
For Honor v10 data, you may want to take a look at [the yaml](https://drive.google.com/file/d/1BbOQuU7Oqt65qUP_xYmw5irrZKcVXy4i/view?usp=sharing).

nvgtriho, 2021-01-08:
> Here is the [link](https://drive.google.com/drive/folders/1OrrODEqNn_0_sJwqaNt-faYWdzJLOhv4?usp=sharing) to the scripts and configuration files for the example data.
> For Honor v10 data, you may want [...]

Thanks. Let me try

closed 2021-01-08 by nvgtriho

minje-KIM, 2023-08-21:
can you share the yaml file link again? 
It expired.

JzHuai0108, 2023-08-21:
I repaired the expired link. Can you open the two links now?

minje-KIM, 2023-08-22:
Yes. The link works now. 
Thanks!