OSUPCVLab/mobile-ar-sensor-logger #11: native crash

yangshoayun, 2021-01-13:
[image]
开始录制视频后报错，请问有什么解决的方法么

yangshoayun, 2021-01-13:
2019-08-26 14:15:55.756 6922-6922/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-08-26 14:15:55.756 6922-6922/? A/DEBUG: Build fingerprint: 'Android/msm8998/msm8998:8.1.0/OPM1.171019.019/xzk08261338:userdebug/test-keys'
2019-08-26 14:15:55.756 6922-6922/? A/DEBUG: Revision: '0'
2019-08-26 14:15:55.756 6922-6922/? A/DEBUG: ABI: 'arm64'
2019-08-26 14:15:55.756 6922-6922/? A/DEBUG: pid: 4112, tid: 4352, name: TextureMovieEnc  >>> com.polytope.dataacquisitionprogram <<<
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x66006600750042
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x0   0000007896cbc380  x1   00000078a60417e0  x2   0000000000630069  x3   0000000000000922
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x4   00000000000fff40  x5   0000000000000000  x6   000000002385a813  x7   00000078973b8b80
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x8   0000000000000002  x9   0000000000000000  x10  0000007897021200  x11  0000000000000000
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x12  0000000000000006  x13  0066006600750042  x14  000000000000812f  x15  0000000000000005
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x16  0000000000008a4a  x17  0000000000000004  x18  0000000000000058  x19  0000007896cc8f58
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x20  0000000000000000  x21  0000007896cbc380  x22  00000078a60417e0  x23  00000078a32bd800
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x24  00000078a32bd800  x25  0000000000000000  x26  0000000000000006  x27  0000000000000001
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     x28  0000000000000000  x29  00000078977250d0  x30  00000078a1a061e4
2019-08-26 14:15:55.757 6922-6922/? A/DEBUG:     sp   0000007897725010  pc   00000078a1a051c8  pstate 0000000020000000
2019-08-26 14:15:55.761 6922-6922/? A/DEBUG: backtrace:
2019-08-26 14:15:55.761 6922-6922/? A/DEBUG:     #00 pc 00000000003641c8  /vendor/lib64/egl/libGLESv2_adreno.so (A5xContext::UpdateTextureSampler(EsxSamplerDesc const*, A5xTextureObject const*, A5xSamplerObject const*)+292)
2019-08-26 14:15:55.761 6922-6922/? A/DEBUG:     #01 pc 00000000003651e0  /vendor/lib64/egl/libGLESv2_adreno.so (A5xContext::ValidateTexSamplersCommon(A5xProgram*, EsxBitField96*)+2260)

JzHuai0108, 2021-01-13:
Thank you for using Mars Logger.

1. Can you please show more backtrace? From the shown backtrace, I can't tell what could be the cause.
2. Have you tried the [release version 1](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/releases/tag/v1.0-android)? Does it have this crash?

JzHuai0108, 2021-05-31:
I will close it for now because the issue is in stasis for too long.

closed 2021-05-31 by JzHuai0108