OSUPCVLab/mobile-ar-sensor-logger #12: Question about readout time in movie_metadata.csv

QMonteCarlo, 2021-03-03:
Hi,Dr Huai:
    I enjoy using your app, and I have installed your app on both Xiaomi mi8 and mi11. 
   One question confused me. For my mi8 android phone, the frame readout time is 23848580ns as reported, and the typical value for exposure time is 1e7 or 2e7 ns，together they are greater than frame duration time 3.3e7 ns。So what is the exact meaning of the frame readout time, I'd like to hear from you.

JzHuai0108, 2021-03-04:
The frame readout time is the time to read out all rows H of an image. For a rolling shutter camera, each row is reset, exposed, transferred, and readout, with a constant delay relative to its previous row, i.e., line delay. The frame rate depends on exposure time and line delay. Hope it clarifies the concept.

QMonteCarlo, 2021-03-05:
Thanks for your reply. Here there are two models: [https://www.researchgate.net/profile/Hyungchul-Yoon/publication/319641499/figure/fig4/AS:537781174378497@1505228431208/Readout-time-and-exposure-time-in-rolling-shutter-in-single-image-frame.png](url) and [https://th.bing.com/th/id/Rd8362b1e603aa66b134acd44cdb7631e?rik=joKMAD2tP0UJbg&riu=http%3a%2f%2fvisionblog.vieworks.com%2fwp-content%2fuploads%2f2020%2f01%2fRolling-Shutter-exposure-and-readout.png&ehk=GO%2fvx9gM3q66Zf0BDF24m3%2bXJZozIR5DdvThffCgPwY%3d&risl=&pid=ImgRaw](url). It seems that you are using the second model? 
I just want to know 1.which decides the line delay, the reset time or the readout time 2.should total rest(or readout) time + exposure time < frame dureation time?

JzHuai0108, 2021-03-05:
In my opinion, the two schematic drawings are essentially the same. The nuance is that the first drawing does not show elapsed times for reset and readout. 
1. The line delay is the relative time offset between consecutive rows. It is not determined by reset time or readout time. You may say line delay is the time offset between consecutive resets or consecutive readouts. You see the two descriptions are the same.
2. I agree with you. Exposure time + frame readout time should be less than 1/frame rate.

QMonteCarlo, 2021-03-05:
Thanks for your quick answer and detailed explanation. Maybe some literatures have confused the line delay with readout time(eg. https://intra.ece.ucr.edu/~mourikis/papers/LiI2014JRR_RS.pdf)? I guess the line delay is not included in your output file, and it should be calibrated out?

JzHuai0108, 2021-03-06:
I honestly think Li and Mourikis are correct, though they do not specifically say that line delay and readout time are related by a factor. 
The output file contains the frame readout time which is line delay X H,
where H is the active array height, typically the image frame height.
You will want to read google's great explanation about rolling shutter skew, frame readout time, at [here](https://developer.android.com/reference/android/hardware/camera2/CaptureResult#SENSOR_ROLLING_SHUTTER_SKEW).

closed 2021-05-05 by JzHuai0108