OSUPCVLab/mobile-ar-sensor-logger #14: The recorded video has wrong aspect ratio

yuweijheng, 2021-04-24:
Hi Dr. Huai:

I am wondering if you can kindly walk me through how the video resolution is set in the recorded file for the Android logger? On my devices, if the aspect ratio is not set to be 16:9, the output file will be squished or stretched into a 16:9 video.

Thanks.

JzHuai0108, 2021-04-25:
Thank you for using MARS logger.

Does your Android phone camera get proper videos with resolutions both 1920 x 1080 and 1280 x 720?
If both resolutions lead to normal videos, then I would imagine that the issue has to do with the Android API.
Otherwise, there could be some issue in the app implementation. Please let me know about it.

As for walking through the resolution setting, I think you already know it :). 
First on the IMU tab, tap the upper right corner Settings, 
then in the Camera Settings list, hit Frame Size, choose the resolution you prefer.
Now go to the video capture tab, you should see that the chosen frame size shown on the top of the screen.

JzHuai0108, 2023-06-30:
I close the issue for dormancy.

closed 2023-06-30 by JzHuai0108