OSUPCVLab/mobile-ar-sensor-logger #15: Rosbag Issue

qamaruislam, 2021-04-28:
Hi,

First of all thank you @JzHuai0108 
[image]
Jianzhu Huai so much for your valuable work. However, while implementing Covert to Rosbag wiki tutorial I encounter the following error in Ubuntu 20.04.2 LTS

python $BAG_PYTHON --video $ANDROID_DATA_DIR/movie.mp4 \
> 
Traceback (most recent call last):
  File "/home/qamar/vio_common/python/kalibr_bagcreater.py", line 10, in <module>
    import rosbag
  File "/opt/ros/noetic/lib/python3/dist-packages/rosbag/__init__.py", line 33, in <module>
    from .bag import Bag, Compression, ROSBagException, ROSBagFormatException, ROSBagUnindexedException
  File "/opt/ros/noetic/lib/python3/dist-packages/rosbag/bag.py", line 51, in <module>
    import yaml
ImportError: No module named yaml

It is so kind of you, if you kindly tell me How to solve this issue.

I am looking forward to hearing from you soon.

Regards

JzHuai0108, 2021-05-03:
Have you tried
```
pip2 install pyyaml
```?

according to [here](https://stackoverflow.com/questions/50868322/importerror-no-module-named-yaml/50868994).

qamaruislam, 2021-05-04:
Dear @JzHuai0108 

Thank you for your valuable comment. I successfully installed pip2 install pyyaml as you mentioned above. But when I run the program and again encounter the other error:

(evaluation) qamar@qamar-SVE14137CVP:~$ BAG_PYTHON=~/workspace/local_install/vio_common/python/kalibr_bagcreater.py
(evaluation) qamar@qamar-SVE14137CVP:~$ ANDROID_DATA_DIR=~/Datasets/Mars_Logger/outdoor_4v
(evaluation) qamar@qamar-SVE14137CVP:~$ python $BAG_PYTHON --video $ANDROID_DATA_DIR/movie.mp4 \
> 
given video time offset 0.0
video frame rate 29.942
#Frames 9367.0 in the video /home/qamar/Datasets/Mars_Logger/outdoor_4v/movie.mp4
video frame index start 0 finish 9366
Traceback (most recent call last):
  File "/home/qamar/workspace/local_install/vio_common/python/kalibr_bagcreater.py", line 476, in <module>
    main()
  File "/home/qamar/workspace/local_install/vio_common/python/kalibr_bagcreater.py", line 376, in main
    videotimerange = write_video_to_rosbag(
  File "/home/qamar/workspace/local_install/vio_common/python/kalibr_bagcreater.py", line 309, in write_video_to_rosbag
    image_np = cv2.pyrDown(image_np, dstsize=(w / 2, h / 2))
TypeError: integer argument expected, got float

Kindly tell me you know How to solve this issue.
[image]

Thank [you]([url]([url](url)))

Ceopee, 2021-05-04:
Hi @qamaruislam,

I have faced the same issue. I guess there is a casting issue there. My workaround is in [line](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py#L309)

`            image_np = cv2.pyrDown(image_np, dstsize=(w / 2, h / 2))
`
to
`            image_np = cv2.pyrDown(image_np, dstsize=(int(w / 2), int(h / 2)))
`

Hope it works for your.
Cheers.

qamaruislam, 2021-05-04:
Dear @Ceopee 

Thank you so much for your valuable comment, Now it works well.

Stay Happy Stay Healthy.

Kind Regards

JzHuai0108, 2021-05-05:
Sorry for the inconvenience. In fact, I had updated the bag creator in another branch. I will update the master branch soon.

Here is my solution.
```
image_np = cv2.pyrDown(image_np, dstsize=(w / 2, h / 2))
```
to
```
image_np = cv2.pyrDown(image_np, dstsize=(w // 2, h // 2))
```

closed 2021-05-09 by qamaruislam