OSUPCVLab/mobile-ar-sensor-logger #18: Question about camera intrinsic parameters

Shane922, 2021-05-28:
Hi, thanks for your excellent work!
I have some questions about the camera intrinsic parameters. As your guidance in wiki, the fx and fy should be set by the reference(fx[px]: 954.7833, fy[px]: 954.7833) in movie_metadata.csv. But when I convert the data recorded to Ros_bag with the scripts kalibr_bagcreater.py and calibrate the intrinsic parameters by using [Kalibr](https://github.com/ethz-asl/kalibr), I find the fx: 834.068 and fy: 853.221, which are different from the data in movie_metadata.csv. So my question is which data should I choose, from movie_metadata or Kalibr ?
The equipment is OnePlus7t, and I set the AF mode to 1 when recording. I also turn on the option " manually control ISO and Exposure" and adjust ISO to 800, Exposure time to 10ms. The frame size I use is 1280x720(Yes, it was rotated to width:1280, height: 720 in kalibr_bagcreater.py).

JzHuai0108, 2021-05-30:
Good point. 
I empirically find the reference focal length in the movie_metadata.csv work well most times. 
On the other hand, if you can calibrate the intrinsics with Kalibr, then I think its results are more reliable. 
Though I am confounded by the large difference between the two focal lengths.

Shane922, 2021-05-31:
> Good point.
> I empirically find the reference focal length in the movie_metadata.csv work well most times.
> On the other hand, if you can calibrate the intrinsics with Kalibr, then I think its results are more reliable.
> Though I am confounded by the large difference between the two focal lengths.

Thanks for your reply, I will close this issue.

closed 2021-05-31 by Shane922