OSUPCVLab/mobile-ar-sensor-logger #19: questions about image records

Dytong-31, 2021-05-31:
Hi, thank you very much for your work in data recording.
I found that the app outputs a video stream. If I want to output a 30hz original image, do I need to modify the TextureMovieEncoder part?
thanks!

JzHuai0108, 2021-05-31:
I am unsure about the question.
Do you want to save the stream as separate images? Then I would advise you against it, because this takes too much time and memory. You can convert the saved video to separate images with the [tool](https://github.com/JzHuai0108/vio_common/blob/master/app/video2images_kalibr.cpp).

Dytong-31, 2021-05-31:
Thank you very much for your reply!
Because I want to input the collected data into the SLAM system. I would like to ask whether additional noise will be introduced if the video is converted into an image? Compared with direct record images as input of SLAM system.

JzHuai0108, 2021-05-31:
The H.264 compression scheme used in saving the video may cause small information loss.
Extracting images from the video, as far as I know, do not cause artifacts, unless the image is down-scaled.
All in all, these nuances have much less impact on SLAM accuracy compared to other factors, e.g., rolling shutter, time synchronization, inaccurate calibration, IMU noises.

Dytong-31, 2021-06-01:
Thanks a lot!
I also want to ask what does "Sensor frame duration[nanosec]" and "Frame readout time[nanosec]" in the movie_metadata.csv mean, do I need to consider them in the SLAM system? And when I click on the screen to fix the focal length, the focal length will still change as:
[image]

JzHuai0108, 2021-06-03:
My understanding is that, depending on the implementation of locking autofocus on an android phone, the focus may still vary a bit even when it is locked.

[sensor frame duration](https://developer.android.com/reference/android/hardware/camera2/CaptureRequest#SENSOR_FRAME_DURATION) is the like the exposure duration.
[frame readout time](https://developer.android.com/reference/android/hardware/camera2/CaptureResult#SENSOR_ROLLING_SHUTTER_SKEW) is the time to readout a frame of a rolling shutter camera.

Dytong-31, 2021-06-04:
Thank you very much.I found that the IMU and image timestamps are very different, should I do something to deal with it? But I think the difference is too obvious. 
[image]
[image]

JzHuai0108, 2021-06-05:
Can you look into edge_epochs.txt?
The file will likely inform that the two sensors use different clocks. 
Since it records the time for both clocks at the start and the end, then you can compensate for the time difference.

Dytong-31, 2021-06-06:
I recorded data on oppo r15 and Huawei mate40. The video captured during horizontal recording is still 1280*720, as:
[image]
so I rotated it by 90 degrees.
The problem is that running vins_mono always shows initialization failure, "solve g failed!". I suspect it is caused by an error in the configuration file.
[image]
[image]

JzHuai0108, 2021-06-07:
1. Please check the unit of td.
2. Make sure the rosbag has correct image size 1280 x 720 with the [tool](https://github.com/JzHuai0108/vio_common/blob/master/python/play_images_in_rosbag.py).

Dytong-31, 2021-06-12:
Thank you! 
I still have some doubts about td.I recorded data on  Huawei P40
The  gyro_accel.csv shows:
[image]

The  movie_metadata.csv shows:
[image]

But the edge_epochs.txt shows as:
[image]
I want to ask why this data 25549036328913 is not displayed in movie_metadata.csv at all, and how should I calculate td? How to understand the Difference shows on the  edge_epochs.txt?

JzHuai0108, 2021-06-15:
Sorry for replying late as I have been toiling for manuscript revisions.
In short for the Huawei P40 case, time offset is not necessary because the camera's time source is TIMESTAMP_SOURCE_REALTIME. td can be initialized to zero in this case.

The explanation in edge_epochs.txt is too wordy :(.
For your earlier data, time offset is needed. td = imu time - camera time.
Based on the screenshots of your data, td ~= (461823700868799 - 218646769700000) / 1e9 = 243176.931168799 s. 
The better estimate of td should have been provided in the corresponding edge_epochs.txt.

closed 2021-08-16 by JzHuai0108