OSUPCVLab/mobile-ar-sensor-logger #20: Commercial use

Jpvmello, 2021-07-13:
Hi,

Congratulations for your work. I see that the repo is under GNU License, but when I download the apk the license agreement show some restrictions. They do not seem to restrict commercial use (except for renting the app or similar stuff), but due to licenses duality I would like to be completely sure: can I use the app for collecting IMU data to internal use for the startup I work in?

Thanks in advance!

JzHuai0108, 2021-07-16:
As a noob to these licenses, I honestly cannot tell them apart.  Before consulting with professor Yilmaz, I personally think it is OK to do so.

JzHuai0108, 2021-07-19:
Apparently, the problem arises from the inconsistency between the [GNU license](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/LICENSE) and the [EULA agreement](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/blob/master/android-mars-logger/app/src/main/assets/EULA).
Honestly, I am not sure about which clauses are conflicting. Can you please look into this, professor?@alpery74

alpery74, 2021-07-19:
It is okay for commercial use but you need to cite the use of the software in acknowledgments of open source libraries.

closed 2021-08-16 by JzHuai0108