OSUPCVLab/mobile-ar-sensor-logger #22: NullPointerException even after editing build.gradle

jxu12345, 2021-11-27:
Hi Josh,

I read the issue on the null pointer exception and attempted the easy fix, deleting all references to the signing config. However, I'm still getting a nullPointerException, except this time with no error message.

These are the edits I made. Please check that I'm doing them correctly.
[image]

This is the error message I get. 
[image]

Thanks,
Jason

JzHuai0108, 2021-11-28:
I recall there was some line in the code causing the problem, and I fixed these issues a while ago.
This repo currently has two apps, one for iOS, one for android. It is very inconvenient for pull requests.
I am thinking to split it into two. I will do this in a couple of days. 
Meanwhile, you may try the bitbucket repo for [android MARS logger](https://bitbucket.org/JzHuai0108/mars_logger_android).

jxu12345, 2021-11-28:
Thank you very much! The code from bitbucket works without problem.

closed 2021-11-28 by jxu12345