OSUPCVLab/mobile-ar-sensor-logger #23: The calibrated time offset are different between Vins mono and Kalibr

mppcasc, 2021-12-24:
Hi, 

I used this app to record the video and IMU data, and then used VINS Mono and Kalibr respectively to calibrate the time offset, the results were very different. I use the calibrated time offset to synchronize the video frame and IMU data, and then use the IMU data to stabilize the video, **Kalibr calibration result was almost correct**, so the result of VINS Mono is wrong?  I checked the configuration file of VINS Mono, all Settings are correct.
What are the possible reasons for the difference between VINS and Kalibr?
In addition, I found that in the video recorded by the app, the first 2 to 3 frames are the same, but the corresponding frame timestamps are different?

JzHuai0108, 2021-12-26:
IMHO, kalibr does offline calibration, so its results are probably more reliable than VINS Mono which estimates time offset on the fly.
The rolling shutter effect may make a difference too.
We have implement rolling shutter camera - IMU calibration on top of Kalibr at [this fork](https://github.com/JzHuai0108/kalibr).
The VINS Mono supports compensating the rolling shutter effect.
After all, the original Kalibr and VINS Mono should have similar time offset with expected difference within say 0.1 s.

How do you tell that the first 2 to 3 frames are exactly the same? Are you using the [conversion script](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py)? It checks that the number of timestamps is same as the number of frames in the video.

mppcasc, 2021-12-30:
Hi, Now the calibration results of Kalibr match those of Vins Mono, although there is still a slight difference, about 2-3ms. The difference should be that the Kalibr itself does not support rolling shutter. 
The [improved version](https://github.com/JzHuai0108/kalibr) of Kalibr you provided should reduce this error, and I will test it later.

> What are the possible reasons for the difference between VINS and Kalibr?

This is because I set IMU parameters incorrectly in Kalibr. Therefore, the results of Kalibr calibration are all wrong.

Using the[ IMU noise parameters](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki#configure-imu-parameters) you provided in wiki, the IMU fit is close to perfect in Kalibr. 

[image]

> How do you tell that the first 2 to 3 frames are exactly the same? Are you using the conversion script? It checks that the number of timestamps is same as the number of frames in the video.

I used the conversion script you provided, and there was a high probability of repeated frames in the first few frames of the recorded video.
I also noticed that frame_timestamps.txt records all frames data while movie_metadata.csv loses the first frame. This happens to the first recorded video data every time you restart the software. I don't know if this is related to the mobile phone I use, I will track this problem later. Thank you again for your work.

closed 2021-12-30 by mppcasc

JzHuai0108, 2022-01-01:
Good job. 

Good observation. Yes. frame_timestamps.txt is the one we are to use for frame timestamps. 
The movie_metadata.csv may miss some lines.