OSUPCVLab/mobile-ar-sensor-logger #24: Caused by: java.lang.IllegalArgumentException: path may not be null or empty string. path='null'

sarimmehdi, 2022-01-26:
I am unable to open your project in android studio as I get this error.

sarimmehdi, 2022-01-27:
The problem was solved by removing the following lines from build.gradle:

```
signingConfigs {
        release {
            Properties properties = new Properties()
            properties.load(project.rootProject.file('local.properties').newDataInputStream())
            storeFile file(properties.getProperty('storeFile'))
            keyAlias properties.getProperty('keyAlias')
            storePassword properties.getProperty('storePassword')
            keyPassword properties.getProperty('keyPassword')
        }
    }
```
and then removing this line from the `release` braces of `buildTypes`:

`signingConfig signingConfigs.release`

closed 2022-01-27 by sarimmehdi

JzHuai0108, 2022-01-29:
Good job. Actually, these lines are related to building the release version which requires your own keystore and key. You know I can't share them with you :). The shortcut is simply to comment out the release build options or to use the debug version.