OSUPCVLab/mobile-ar-sensor-logger #26: About imu frequence is not accuracy

jingxizc, 2022-07-12:
thank you, really an nice job!
my test device is an android moble phone ,  asus rog phone 5s...
I use you release apk get a nice result, at DELAY_FASTEST mode, i can get 500hz result, at DELAY_GAME mode, get 50Hz，
when i want to use your code in my application to get 500hz imu, i failed , at DELAY_FASTEST, i can only get 50hz...
I don't konw why...please help me!
I just copy your IMUManager.java, no any modified!

JzHuai0108, 2022-07-12:
Thanks for using mars logger.
I think you should double check that the value of DELAY_FASTEST is passed to the IMU Manager.
Try print the value of mSensorRate at [here](https://github.com/OSUPCVLab/marslogger_android/blob/837abff3de25bc447b2451677e57979f2e98417e/app/src/main/java/edu/osu/pcv/marslogger/IMUManager.java#L226-L227).

jingxizc, 2022-07-15:
@JzHuai0108 Thanks a lot !  I found that stupid error, as you can see, I forget to modify the mSensorRate value......

closed 2022-07-15 by jingxizc