OSUPCVLab/mobile-ar-sensor-logger #28: 关于将采集的数据打包问题

zhangzuhao111, 2022-09-14:
您好，我刚开始接触camera+imu融合视觉学习。
        目前已经使用了您的apk进行了数据的采集，在本地查看数据，录制的视频是MP4格式，我参照网上的rosbag内的数据集，将视频格式根据帧数转成了一帧帧的图像，图像数量与 frame_timestamps.txt 这一文本文件中时间戳也对应，也就是说该文本文件的时间就是对应的每一帧图像的时间呢？
        谢谢！期待您的答复！

JzHuai0108, 2022-09-15:
你是对的。“该文本文件的时间就是对应的每一帧图像的时间”。：）

zhangzuhao111, 2022-09-17:
您好，感谢您上次的回复！
我参照了您的打包格式，整理好了dataset文件，里面包含了单目相机图像（cam0文件夹）和imu.csv，并且按照格式将时间戳标注在了每一帧image上。
通过kalibr_bagcreater.py 脚本进行打包，我选择了--folder [dataset]这一方式，结果打包后文件出现了问题（文件大小仅为4.1kb，这显然不对），我目前还在查看源代码，想知道我的文件格式是否有问题。
此外，我在您的wiki上找到了您之前录制并打包好的movie.bag，并且想测试一下能否在vins-mono环境中运行，结果也出现了问题，目前还不清楚原因。
希望能得到您的简单解答！这或许对我有很大帮助，感谢！

JzHuai0108, 2022-09-17:
你可以将时间戳标注到每帧图像上然后再打包，也可以直接用kalibr_bagcreater处理mars logger录制的数据。为了看清出的问题，你帮忙截取一下打包的屏幕结果？
对VINS Mono，你得截取一下屏幕？

zhangzuhao111, 2022-09-17:
感谢您这么快速的回复。
我通过查看kalibr_bagcreater.py源代码发现，确实可以直接处理mars logger录制的视频流信息，这个以后我会去尝试的。
我分享一下我的截图：
[image]
上面的是我打包之前的文件布局。
[image]
上面是我标注好时间戳的image。
[image]
这个是终端运行的情况。
[image]
运行以后会发现确实能输出一个打包好的文件，但是发现只有4kb，这肯定是不对的
我猜测或许是imu.csv的格式有问题？下面是我的imu.csv的信息
[image]

我想这样描述应该准确了吧，谢谢！
还希望您不吝赐教

JzHuai0108, 2022-09-17:
1. 建议用kalibr_bagcreater直接处理mars logger录制的视频流信息, 这样会简便许多。
2. 为啥IMU.csv的时间戳很多重复？
3. 你是用的https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py？

zhangzuhao111, 2022-09-17:
谢谢您的答复
1.那我就去尝试处理视频流的信息方法
2.由于保存csv的时候不小心点了确定导致的，这是我的疏忽！抱歉
3.对的，但是现在我还不清楚处理视频流所需要的的格式

zhangzuhao111, 2022-09-17:
我想问一下，视频流打包是否有像图像打包一样的直观说明呢，类似于：https://github.com/ethz-asl/kalibr/wiki/bag-format，这个网址中提供的一样，谢谢

JzHuai0108, 2022-09-17:
你看一下[这个](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py#L128-L133)？和[这个](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki#convert-to-rosbag)?

zhangzuhao111, 2022-10-06:
您好，感谢您的耐心答复！
接上次的问题，我最终使用了kalibr_bagcreater直接对mars logger录制的视频流信息进行打包，效果很好。由于没有进行相机标定，所以vins-mono下的rviz可视化轨迹很飘。
所以我必须对相机和imu进行标定。遇到了几个问题还想请教您一下。
1.还是之前的那个问题，如果我想对图像和imu进行打包，我的想法是对mp4视频流进行按帧数裁剪，这里我发现直接将视频裁剪完得到的image数量和本地的image时间戳无法对应上（要么image数量多于时间戳数量，要么时间戳数量多于image数量），但差距不大（通常差距为5以内）。所以我找到了一个两者比较接近的数量样本，将最后多余的几张image直接删掉了，不知道是否可行？
2.我按照您给出的标准格式进行文件布置，如下：
[image]
[image]
输入命令行：./kalibr_bagcreater --folder dataset-dir/. --output-bag awsome.bag，得到如下的提示：
[image]
[想知道提示中为什么只显示了将图片写入rosbag中呢？imu数据没有看到，或许我可以利用解包查看一下，但我没有找到](https://github.com/ethz-asl/kalibr/wiki/bag-format#bagextractor)

还希望您不吝赐教，谢谢！

JzHuai0108, 2022-10-06:
确认一下kalibr_bagcreater, 在vio_common repo中，还是在kalibr repo中？我建议用vio_common中的[bagcreater](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki#convert-to-rosbag). kalibr repo中的bagcreater可能不支持mars logger录的IMU数据格式。

zhangzuhao111, 2022-10-15:
您好，我重新整理了一下kalibr_bagcreater，确认是来自vio_common repo中的。
但是目前依然没有解决利用kalibr_bagcreater单独打包imu的问题。
[image]
错误提示是告知我必须要放入一个image或mp4，但是加入后bag文件会变得很大，导致内存超出。
请问kalibr_bagcreater能否单独打包imu数据呢？
万分感谢？

JzHuai0108, 2022-10-16:
kalibr_bagcreater in vio_common可以单独打包imu数据，只要改一下
[这一句](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py#L569)为
```
else
    pass
```

zhangzuhao111, 2022-11-06:
您好，在最后的imu+cam联合标定过程中，我看到了imu的配置参数已给出参考，[https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki#configure-imu-parameters](url)，所以我是直接采用了所给的参数。
[image]
对比我自己实验得到的数据来看，二者差别较大，使用所给参数其轨迹好很多，但是仍然会飘。请问如何改进或者根据自己的实验结果来配置呢？或者说在imu标定过程中有什么好的建议呢？
非常感谢！

zhangzuhao111, 2022-11-06:
[您好，在最后的imu+cam联合标定过程中，我看到了imu的配置参数已给出参考，，所以我是直接采用了所给的参数。](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki#configure-imu-parameters)

[image]
[image]

对比我自己实验得到的数据来看，二者差别较大，
使用所给参数其轨迹好很多，但是仍然会飘。请问如何改进或者根据自己的实验结果来配置呢？或者说在imu标定过程中有什么好的建议呢？
非常感谢！

JzHuai0108, 2022-11-07:
你的帖子中开头说在做imu+cam联合标定，中间在用vins mono做imu+cam 里程计，最后又说imu标定。请问你是在做啥？想解决什么应用中的IMU参数问题？

zhangzuhao111, 2022-11-08:
您好，或许是我表达的有问题，也可能是我对vins-mono的标定理解不好。
我对单目相机+IMU的标定理解是：
1.首先对camera进行单独位姿标定，得到内外参和畸变参数。
2.然后对imu进行单独标定，得到加速度计和陀螺仪的噪声参数。
3.最后把上面两次标定会得到配置文件cam-chain.yaml和imu.yaml，利用kalibr的kalibr_calibrate_imu_camera脚本进行标定，得到最后的cam to imu的位移、旋转矩阵。
写入vins-mono的config文件中，进行测试。

这里我遇到的问题就是：在写vins-mono的配置文件中，还是需要写入imu的参数？
[image]
但是在我第二步的时候，我就已经通过imu单独标定得到了呀
[image]
但是，如果我用第二步得到的参数写入进去的话，效果很差；若是用您给的imu配置参数，效果会有很好的提高。我不是很清楚这是什么原因。
[image]
希望能得到您的解答。非常非常感谢！

JzHuai0108, 2022-11-08:
There are two points to consider when you are setting the IMU noise parameters for VINS-Mono, from those values provided by the Allan variance analysis.
1. The VINS-Mono noise parameters should be the Allan variance values enlarged by a factor \sqrt(f) because of its integration scheme. see also [the text below the table](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki). f is the IMU rate.
2. If necessary, it may be necessary to further enlarge these values to account for unmodeled noise effects. For instance, white noise std X 5, bias random walk std X 10. But in this case, I feel that the allan variance values after scaled up by \sqrt(f) look good for VINS Mono.

zhangzuhao111, 2023-03-25:
您好，我想问一下仓库中提供的apk，在哪里查看imu的频率呢？也就是标定需要的yaml文件中的：
[image]

JzHuai0108, 2023-03-25:
android app的右上角有一个三个点的设置按钮，进去之后就可以看到了。

zhangzuhao111, 2023-03-25:
[image]
抱歉，我好像没找到那里可以看到具体的数值。
此外我想问一下这四个按钮是控制频率的吗？

JzHuai0108, 2023-03-25:
是的。安卓提供的设置IMU频率的接口是按这四个选项频率来设置的。

zhangzuhao111, 2023-03-25:
感谢！那请问具体的数值有吗？

JzHuai0108, 2023-03-25:
一般Game 50 或100 Hz, Fastest 100或200 Hz.

JzHuai0108, 2023-06-30:
I close this issue for dormancy.

closed 2023-06-30 by JzHuai0108