OSUPCVLab/mobile-ar-sensor-logger #30: 关于相机自动对焦和参数设置的疑惑。

risingryu, 2022-12-13:
首先，十分感谢您能分享这个优质的软件。我的疑惑是来自于，想要进行摄像头的标定。从实时的图像发现焦距一直在改变。按照wiki的方法点击屏幕，焦距还是在一直变化。我觉得焦距一直在变化的话内参没有办法标定。另一个疑惑是看到其他的issue里有询问如何改变分辨率和采样频率。您给的回复是在右上角的设置里面进行改变，但是我没找到相应的设置界面。因此想向您请教一下。

JzHuai0108, 2022-12-13:
谢谢提问。
1.图中那个是焦距。
2.一般点击屏幕之后, 焦距应该是固定的了.如果不是,能说一下手机型号和所用的marslogger的安装源，我来debug一下。
3. 参数设置的界面不是特别友好哈。在初始的界面底部有个导航栏，有IMU，视频等，切换到IMU界面，右上角有三个点的按钮，进入可以手动设置焦距等。

risingryu, 2022-12-13:
非常感谢您这么快回复。我在华为平板上是可以固定焦距了。不能固定焦距的手机型号是realmeX2pro 安卓版本是安卓11，下载是按照github教程下载的release版本的apk  [https://github.com/OSUPCVLab/mobile-ar-sensor-logger/releases/download/v1.0-android/app-fdroid-release.apk](url) 。但是我还是没有找到设置的界面。您说的那个导航栏是在图片上的界面吗？初始的界面是指刚打开软件，还是刚点下录制按钮的时候吗？（我试了都没找到～）
[image]

JzHuai0108, 2022-12-13:
1. 华为的平板Gyro数据读不到，平板的型号是huawei MatePad BAH3-W09 HarmonyOS2.0（手机是正常的，给您反映一下）.他有可能没有Gyro.
2. 焦距的设定是手动触屏定的,触屏近的物体焦距会较小,触屏远的物体焦距会稍大. 但是实际物理焦距手机才4mm, 所以焦距变化很小(从聚焦无穷到聚焦3cm).
3. fastest, game etc 跟手机实现有关。一般game 50 Hz, fastest 100 or 200 Hz. 详情看一下[这里sensor_delay_game](https://developer.android.com/guide/topics/sensors/sensors_overview).

LiuSnake7788, 2023-04-09:
作者你好，同样十分感谢您的app，我在使用时同样遇到了一些问题。
我使用的HUAWEI P50手机似乎在录制视频时会自动变焦，该如何设定为焦距不变？我看2.0的app说明中有一些手机可以锁定焦距。

JzHuai0108, 2023-04-09:
试一下单手指长按屏幕三秒钟？一般来说焦距就固定了。

LiuSnake7788, 2023-04-09:
已经解决了！我误解了在Setting调节ISO和曝光时间的用处，原来是长按后才会应用我输入的设置，现在问题已解决
十分感谢！

JzHuai0108, 2023-06-30:
I will close this issue for dormancy.

closed 2023-06-30 by JzHuai0108