OSUPCVLab/mobile-ar-sensor-logger #31: 对于地磁传感器数据的问题

chen579, 2023-05-22:
希望作者大大能够再推出一个能导出地磁数据的版本，第二个版本中数据包含地磁数据但是导出的数据并没有，非常感谢，好人一生平安

JzHuai0108, 2023-05-26:
我好像没有保存地磁数据，但这是个比较简单的功能。我有时间尽早在代码里实现一下。如果你现在可以从源码安装，你可以试着添加一下，照着现有的加速度计的保存方法，可以保存到一个单独的文件中。

chen579, 2023-05-31:
您好，参考您的意见我这几天也在尝试用AS进行调试，但是遇到了一些没有预料到的问题，在网上寻找的教程也没有帮助，因此可能还是需要您的帮助，希望您能够抽出时间来解决这个问题，可有偿，非常感谢。再次致上诚挚的敬意，非常感谢。

JzHuai0108, 2023-06-01:
1. 你已经成功从源码安装mars logger么？我准备更新一下master, 然后你后面安装一下就可以了。
2. 我需要明确一下功能需求：保存一下地磁数据，对么？
3. 你那边啥时候需要用这个功能？

chen579, 2023-06-01:
您好，目前的情况是这样的
1、我用AS进行调试，在解决了私人key的问题后遭遇了SDK版本不匹配的报错，但是参考网上的教程并没有起到什么作用
2、对，希望能够类似加速度计和陀螺仪的数据形式，最好加速度计、陀螺仪和磁力计的时间戳能够保持一致并保存在同一个csv文件之下
3、希望近期一周左右能够用上
非常感谢您的及时回应，不胜感谢，本人的vx号为18284360471，如果有需要的话可以随时联系，非常感谢

JzHuai0108, 2023-06-01:
1. 在这发个截图？
2. 我现在的想法是把accel, gyro 放在原来的那个文件中，magnetometer 新建一个txt保存，含时间戳。这三个数据用的同一个时钟源。1秒之内三种数据的个数不一定一致。这样可以么？
3. 应该可以搞定。

chen579, 2023-06-01:
好的，具体的情况如下
1、
[image]
2、最好是能够做到这三个数据在时间戳上对齐，因为我希望能够利用这三种数据同时进行惯性位姿的解算，时间戳不对齐的话可能会对最后的结果造成一定的影响
3、非常感谢

JzHuai0108, 2023-06-01:
1. 信息不完整，无法判断问题和原因。
2. 我试一下。

chen579, 2023-06-01:
嗯嗯，麻烦了，非常感谢

JzHuai0108, 2023-06-09:
I finally squeezed in the task to log the magnetometer data. Hopefully this update is in time for your project.
Right now, the attached app logs magnetometer data along with synced accel and gyro data.
The code is at the branch develop on marslogger-android.
The attached apk can be installed by 
```
adb install app-fdroid-release.apk
```
[release.zip](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/files/11702690/release.zip)

chen579, 2023-06-09:
好的，非常感谢。我刚刚尝试了一下，导出的数据包括了加速度计、陀螺仪与磁力计信息，再次非常感谢，我刚刚尝试点击了app中的donate按钮，但是显示无浏览器打开网站，您可以通过我留的vx号和我联系，再次非常感谢

JzHuai0108, 2023-06-12:
Good luck! No need to pay me, just pay the favor forward to others. I will close the issue for now.

closed 2023-06-12 by JzHuai0108