OSUPCVLab/mobile-ar-sensor-logger #33: record game rotation vector

CanCanZeng, 2023-06-30:
Hi, thank you for sharing this tool!
I compiled android version successfully and it worked well. But I'm not familiar with android coding, how can I record game rotation vector at the same time? Can you add this sensor or give me some instruction?

JzHuai0108, 2023-06-30:
A while ago, someone wanted to record the magnetometer data, and I added this feature as done [in this commit](https://github.com/OSUPCVLab/marslogger_android/compare/main...develop).
It should be quite similar to recording the [game rotation vector](https://developer.android.com/guide/topics/sensors/sensors_position#sensors-pos-gamerot) following the android doc.
You already had a good start by succeeding in compiling the mars logger.

CanCanZeng, 2023-06-30:
Thank you for your answer, I will try it later.

closed 2023-06-30 by CanCanZeng