OSUPCVLab/mobile-ar-sensor-logger #34: Can not find the record file

elevenjiang1, 2024-02-18:
Hello, thanks for your work!
However, I am using Xiaomi 14, but I can not find output file in /Android/data, and My phone has no sdcard, so, where may it be output?

JzHuai0108, 2024-02-20:
I recall the logger reports the output path in some screen. Do you notice it?

elevenjiang1, 2024-02-20:
Yes, I see the output folder, but I still can not search it out

[image]

JzHuai0108, 2024-02-20:
Do you see the edu.osu.pcv folder on your phone, as shown below?

[image]

Have you granted the storage permission to the app?

JzHuai0108, 2024-02-20:
[image].

elevenjiang1, 2024-02-20:
Maybe because Xiaomi do not give the root permission to /Andriod/data, some links found like: 
https://web.vip.miui.com/page/info/mio/mio/detail?app_version=dev.20051&postId=37583191

I have chosen another open source projcet: https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors?tab=readme-ov-file, and it saves the data to "DCIM/OpenCamera" path, so that I can find the data.

But still thanks a lot for your help~~
By the way, I also graduate from WHU~~~

closed 2024-02-20 by elevenjiang1