OSUPCVLab/mobile-ar-sensor-logger #35: Question about Timestamp Conversion between App Logs and Real-World Time

CloudFan-cyf, 2024-04-09:
Hello, thank you for your excellent work! I am currently using your application to record data from my phone's camera and IMU. Subsequently, I employ VINS-Mono method to generate a trajectory of the phone's movement (denoted as trajectory A). This is compared against a user trajectory captured and generated by a CCTV camera (denoted as trajectory B) to verify the accuracy of the tracking algorithm I am researching.
My issue arises because the timestamps for each frame of trajectory B are in real-world time (e.g., one frame corresponds to 2024/4/9/9:30), whereas the timestamps recorded by your application (e.g., one frame at 328,551,511,918,592 nanoseconds) do not correspond to real-world time. And timestamps recorded in gyro_accel.csv and fram_timestamps.txt is different,which should I choose?This discrepancy prevents me from synchronizing trajectories A and B, making it challenging to cross-validate the two trajectories.
Could you please explain how the timestamps recorded by the application (for trajectory A) are calculated? And how do they correlate with real-world time?Thanks in advance.

SkyPoet, 2025-05-08:
Do you know what format the timestamp of this software is in? I'm also having issues with timestamp synchronization. Thanks.

SkyPoet, 2025-05-08:
I really appreciate your quick response.
a. After comparing the file you sent with mine, I found that I was using an old version of MarsLogger.
b. Yes, I am using the Android version.

For frame_timestamps.txt
```
Frame timestamp[nanosec]
90274527031000
90274560319000
90274593607000
90274626895000
90274660183000
90274693471000
90274726759000
90274760047000
90274793335000
90274826623000
90274859911000
90274893199000
90274926823000
90274960111000
90274993399000
90275026687000
90275059975000
90275093263000
90275126527000
90275159815000
90275193055000
90275226343000
90275259631000
90275292919000
```
For gyro_accel.csv,
```
Timestamp[nanosec]    gx[rad/s]	 gy[rad/s]	 gz[rad/s]	 ax[m/s^2]	 ay[m/s^2]	 az[m/s^2]
9.02746E+13	0.41813695	0.24388777	-0.008552107	0.8554341	9.720842	0.72442704
9.02746E+13	0.40225446	0.23655738	-0.02565632	0.84107715	9.773484	0.6239285
9.02746E+13	0.38148504	0.229227	-0.042760536	0.8219346	9.859626	0.55692947
9.02746E+13	0.35705048	0.21456626	-0.05497783	0.78364944	9.926624	0.48993045
9.02746E+13	0.3252855	0.19868377	-0.06597339	0.7070791	9.974482	0.41336012
9.02746E+13	0.29596397	0.18157956	-0.07574723	0.654437	10.012766	0.34636107
9.02746E+13	0.26419902	0.1632536	-0.07941242	0.58265233	10.04148	0.28414768
9.02746E+13	0.22754712	0.14126249	-0.08185588	0.51086766	10.084551	0.24586253
9.02746E+13	0.19456044	0.119271345	-0.08307761	0.44865423	10.089336	0.20279172
9.02746E+13	0.15668681	0.0923933	-0.08185588	0.38644084	10.1084795	0.20279172
9.02746E+13	0.120034926	0.07040217	-0.07941242	0.33379877	10.122836	0.18843478
9.02746E+13	0.08460477	0.045967575	-0.073303774	0.28115666	10.141979	0.14536397
9.02746E+13	0.049174614	0.022754714	-0.06597339	0.23808585	10.141979	0.1692922
9.02746E+13	0.018631376	0.001985311	-0.05986475	0.19980069	10.18505	0.1692922
9.02746E+13	-0.010690134	-0.017562363	-0.0537561	0.17108682	10.213763	0.18364914
9.02746E+13	-0.040011644	-0.035888307	-0.04886918	0.14237295	10.232906	0.20757736
9.02746E+13	-0.0632245	-0.05299252	-0.040317077	0.12801601	10.2520485	0.25064817
9.02746E+13	-0.08155045	-0.067653276	-0.034208428	0.11365908	10.280763	0.28893334
```

JzHuai0108, 2025-05-08:
To convert the csv files to a rosbag, please refer to [kalibr_bagcreator](https://github.com/JzHuai0108/vio_common/blob/master/python/kalibr_bagcreater.py).

If you demand high timing accuracy, you may want to smooth the host time with the sensor time by the timestamp corrector as demoed in [livox phone bagcreater](https://github.com/JzHuai0108/vio_common/blob/master/python/lidar/livox_phone_bagcreater.py). To compile the timestamp corrector, follow these [instructions](https://github.com/JzHuai0108/vio_common/blob/master/timestamp_corrector/README.md).

SkyPoet, 2025-05-08:
Thank you for your reply. I would like to ask whether the **timestamp corrector** supports **Ubuntu 16.04** and **Python 2.7**, since I previously deployed VINS-Mono in that environment. If it is not supported, I will upgrade to **Ubuntu 18.04** and **Python 3**. Thanks again! @JzHuai0108

SkyPoet, 2025-05-08:
Thank you very much for your reply. With your help, I have successfully packaged it into a bag and run it in VINS-Mono. 
However, there is an issue: after packaging the data into a bag file using kalibr_bagcreator.py, the timestamps inside are **not Unix timestamps** (i.e., the timestamps in the second column of frame_timestamps.txt), but rather those from **the first column**. Is there a way to use Unix timestamps for packaging instead?
Thanks again! @JzHuai0108

JzHuai0108, 2025-05-08:
I think you are competent enough to tweak the code in kalibr_bagcreator a bit to use the unix time instead of the smartphone uptime.

SkyPoet, 2025-05-09:
Thank you for your response. I now understand what needs to be done. Your reply has been very helpful to me as a beginner.